stuartpittaway/diyBMSv4 #132: Change Ina228 to ina229

Michi202020, 2022-12-14:
Hello Stuart,

I found in Texas instruments the Ina229 and I have buy 30pcs.
I see that the code is the same but it's only the different i2c to SPI. Can you help me for the software that you can change the connection.
The Hardware I can change it self.
I thinks this part is easyer for a alternativ for the 228.

Many Thanks and Best Regards
Michel
[image]

Michi202020, 2023-01-03:
Hello,

if its possible that one person can me help withe the code to change the i2c to spi?
The code for the Ina229 is the same as Ina228 only the communication is the different.

Many Thanks and best Regards
Michael

stuartpittaway, 2023-01-03:
Hi @Michi202020, I'm going to start work on this very soon.
Stu.

Michi202020, 2023-01-05:
Hello Stuart,

Many thanks and we can write if you are starting withe this project.

Best Regards
Michael

stuartpittaway, 2023-03-06:
The changes for the ina229 have now been released, check out my YouTube video for information.

closed 2023-03-06 by stuartpittaway

Michi202020, 2023-03-06:
Hello Stuart, many thanks for your work. I will check this in the next days.
Best Regards
Michael