stuartpittaway/diyBMSv4 #134: When the cell temperature is negative, the rule does not work.

dolnet, 2022-12-18:
When the cell temperature is less than 0* the less than rule stops working which allows you to kill batteries that cannot be charged in sub-zero temperatures, e.g. LIFePO4

fhorst1, 2022-12-19:
Nice find!

Not sure if it would require a fix, as no one should use negative values
anyways.
+5 to +45c is the optimal charging temperature for LiFePo4.

Below 0 (celcius) will be damaging for the cells, as Logic result, the
temperature will never be set to negative values anyways.

In new versions of the DIY BMS that can support new types of batteries who
can withstand below freezing temperatures...
This is something that need to be fixed.

For now, I googled and tried to find comercial available lithium battery
that can withstand below freezing (and charge at those temperatures) and
wasn't able to find any.

Many that have some heating options already built in, bit that's just
keeping the battery above freezing.

If you do know a type that can withstand....
Please let us know!
I'm sure for many people that will be a good solution.

For me in Thailand, I'm more focused on the +45c :-)

dolnet, 2022-12-19:
I see you didn't understand the problem.

I do not want to operate the battery in sub-zero temperatures.
I report a problem when the temperature drops below zero and then the protection associated with too low temperature stops working.

I have set in the rules - temp <3 degrees Celsius disconnect charging, disconnect balancing.
In the range of 0 <> 3 it works correctly, when the temperature drops below 0 the rule allows charging and turns on the relay.

fhorst1, 2022-12-19:
Paweł Cisowski, thank you for your explanation!

Indeed, I understood wrong.
Yes, I understand why this is a serious issue.

Thank you for your explanation!

stuartpittaway, 2022-12-19:
Hi @dolnet , thanks for the bug report.

This is fixed in the upstream code now, but will be part of the next release which includes the PylonTech battery stuff.