stuartpittaway/diyBMSv4 #135: Negative temperatures in rules

overkillthemighty, 2022-12-19:
I was looking at the interface today and saw that the over-temperature rules for modules and batteries had been set off. I checked the main page and saw that it was -2*C. I tired setting the minimum temp for the cells and modules to -5 to see if it changed anything but I was informed 0 was the lowest I can set. I am guessing the rules are being handled different from the actual temp values and the bit overflow is showing up as over temperature rather than a negative.  
I have the v4.40 modules.  
Here is the version info from the interface:
Platform & Version
Host name:[DIYBMS-00048000](http://diybms-00048000/)

Processor: ESP32

Version: [2f3bae13d3bec3e89d065fea34a35898e82bf7aa](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/2f3bae13d3bec3e89d065fea34a35898e82bf7aa)

Compiled: 2022-07-06T10:04:54.852Z

Language: en

SDK Version:v4.4.1-1-gb8050b365e

Min free Heap:93532

Free heap:161568

Heap size:294516

overkillthemighty, 2022-12-19:
I just noticed this is for the older units, I should be in the ESP32 section. You can close this!

stuartpittaway, 2022-12-19:
There is a duplicate bug for this raised earlier today (already fixed in the next release)

closed 2022-12-19 by stuartpittaway