stuartpittaway/diyBMSv4 #141: pack under voltage trigger while ESP32 Cotroller reboots

rolfbartels, 2023-01-21:
When we startup or restart the controller while the system is waiting for all the modules to report in, it's trigering the pack under voltage rule, I assume it's calculating the pack before all cells have responded

stuartpittaway, 2023-01-21:
Can you give me a list of the DIYBMS components you are using? Version of hardware and code firmware version?

rolfbartels, 2023-01-21:
Processor: ESP32

Version: [e843bcb23f6aa0b0b75df78a05d59a0f5dc46bc7](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/e843bcb23f6aa0b0b75df78a05d59a0f5dc46bc7)

Compiled: 2023-01-03T10:07:59.024Z

Module version 421 / [47fe99bc](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/47fe99bc)