stuartpittaway/diyBMSv4 #144: a question regarding the wiki

moscito1010, 2023-01-26:
Hi Stuart
You describe the node red flow in the wiki and show a picture of it. Can I also download the flow somewhere?

stuartpittaway, 2023-01-27:
Hello @moscito1010 , I don't have a wiki or use node red!

Was this published by a DIYBMS user?

moscito1010, 2023-01-29:
from here
https://github.com/stuartpittaway/diyBMSv4/wiki/Visualize-data-from-the-diyBMS

stuartpittaway, 2023-01-30:
Ah I see - thats user generated content by "donnib" - I don't use NodeRed.

Stu.

closed 2023-01-30 by stuartpittaway

stienman, 2023-01-30:
The nodered flow is in the article, hidden in a disclosure labeled
"►Details", immediately above the diagram.  Click "► Details" and it'll
show you the flow code, which I'll include below for convenience.

[
    {
        "id": "d737058.bd3b5f8",
        "type": "comment",
        "z": "efe93325.55893",
        "name": "Push cell data from MQTT to InfluxDB",
        "info": "",
        "x": 300,
        "y": 270,
        "wires": []
    },
    {
        "id": "4146fd37.aa7214",
        "type": "comment",
        "z": "efe93325.55893",
        "name": "Push diyBMS status from MQTT to InfluxDB",
        "info": "",
        "x": 320,
        "y": 375,
        "wires": []
    },
    {
        "id": "bb75249b.baa748",
        "type": "comment",
        "z": "efe93325.55893",
        "name": "Push diyBMS relay status from MQTT to InfluxDB",
        "info": "",
        "x": 330,
        "y": 495,
        "wires": []
    },
    {
        "id": "d2e3da5e.46cb48",
        "type": "influxdb out",
        "z": "efe93325.55893",
        "influxdb": "a6fdab90.b072d8",
        "name": "",
        "measurement": "",
        "precision": "",
        "retentionPolicy": "",
        "database": "database",
        "precisionV18FluxV20": "ms",
        "retentionPolicyV18Flux": "",
        "org": "home",
        "bucket": "diybms",
        "x": 910,
        "y": 675,
        "wires": []
    },
    {
        "id": "5f31136b.6f802c",
        "type": "comment",
        "z": "efe93325.55893",
        "name": "Push diyBMS rules status from MQTT to InfluxDB",
        "info": "",
        "x": 330,
        "y": 765,
        "wires": []
    },
    {
        "id": "60dc208.35703e",
        "type": "change",
        "z": "efe93325.55893",
        "name": "relay2",
        "rules": [
            {
[... 684 lines cut ...]
        "url": "http://influxdb:8086",
        "rejectUnauthorized": true
    },
    {
        "id": "d77c97be.ca8e08",
        "type": "mqtt-broker",
        "name": "Mosquitto",
        "broker": "10.10.0.9",
        "port": "1883",
        "clientid": "",
        "usetls": false,
        "compatmode": false,
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "willTopic": "",
        "willQos": "0",
        "willPayload": ""
    }
]