stuartpittaway/diyBMSv4 #146: Polarization

marcin11, 2023-02-20:
Jlc and I are sort of confused about polarization of D2 diode on 4.5 module. PCB had markings K and + on the same site. According to schematics Cathode should be on the right site connecting directly to F1.
The other question is about D1 on controller.
Should pin 1 Anode be on the bottom right connecting to R17?
Thank you for your hard work.

stuartpittaway, 2023-02-20:
D2 is a SMBJ5.0A TVS DIODE - data sheet https://datasheet.lcsc.com/szlcsc/1912111437_AnBon-SMBJ5-0A_C440263.pdf

The STRIPE of the DIODE should be pointing to the RIGHT of the module PCB (towards positive).

[image]

D1 is a 3 pin device, so no problem with orientation.  Do you mean D3 (next to R17) as per the silk screen, the positive is on the right hand side.

[image]

marcin11, 2023-02-20:
Thank you for your respond.
2nd part of my question referred to D1 on the controller board

[image]

marcin11, 2023-02-20:
Thank you. Please excuse my ignorance. There is 8 or 10 versions of  ATTINY1624 XF, SSU,SSF ... are they interchangeable?

stuartpittaway, 2023-02-22:
They all have differences, but as long as it has the same 14 pin layout it should be fine