stuartpittaway/diyBMSv4 #154: Polarization of D2, D3 and D4

addame, 2023-04-20:
Hi,

jlcpcb is asking to confirm the polarity of D2, D3 and D4 (see bellow the polarities jlcpcb wants to be confirmed). 
For D2 and D3, I have seen that an answer is already given (issue #146): the PLUS pointing to the RIGHT of the PCB. 
Is the PLUS of D4 also pointing to the right of the PCB?

Thank you

[image]

floreseken, 2023-04-20:
They asked me the same thing, but only about D2. I asked them to reverse D2, to match the plus with the plus.

But that was the wrong decision, got the boards with D2 in the wrong direction.

Luckily this this the easiest part to desolder and reverse...

stuartpittaway, 2023-04-21:
@addame as @floreseken pointed out, the rotation should be correct as you have it in the picture.

JLC can't seem to get their head around D2 being a TVS diode - or their stock system is setup incorrectly.

addame, 2023-04-21:
@floreseken and @stuartpittaway thank you for the clarification!

closed 2023-04-21 by addame