stuartpittaway/diyBMSv4 #157: can i replace this parts for the 440 module?

moscito1010, 2023-05-17:
This parts are not availible

i habe change 
D1 to https://www.lcsc.com/product-detail/Voltage-References_Diodes-Incorporated-AZ431LANTR-G1_C460576.html
and
D4 to https://datasheet.lcsc.com/lcsc/1806151820_Hubei-KENTO-Elec-C2297_C2297.pdf

is this ok for D1 ?

stuartpittaway, 2023-05-18:
I've run into problems previously with different voltage reference chips like those, it will work, but sometimes I saw random values

moscito1010, 2023-05-18:
This is bad, can you recommend an alternative? I want to replace all 1.9f with the original v440. Unfortunately, the two parts are currently not in stock at jlcpcb.

stuartpittaway, 2023-05-18:
Can you take the parts off the old boards?

moscito1010, 2023-05-18:
No, I don't have the 1.9f boards anymore. 
No problem, I'll wait until the pcb forge has the components back in stock or buy them separately. The LED should go, it just has a different color if I saw it correctly.

moscito1010, 2023-06-15:
Hi
Other pcb but the same problem :-)
The part AZ432ANTR-E1 is not longer in stock and manufactured. Can anyone tell me a alternative part please.

atanisoft, 2023-06-15:
C460576 is a direct replacement per the manufacturer and is in stock with JLC.

stuartpittaway, 2023-06-16:
Thanks @atanisoft.  @moscito1010 just to note that the part suggested "C460576" should work, its 1.24V rather than 1.25V output.

So you the value you use to calibrate the cell modules may be a little higher.