stuartpittaway/diyBMSv4 #158: Can the v4.5 modules work with the ESP8266 controller?

floreseken, 2023-05-22:
Can the v4.5 modules work with the ESP8266 controller?

I made this combination because of the simplicity of the old controller.

I flashed the controllers and the modules with the latest pre-compiled HEX files I could find in the releases.
But my setup does not work now, and I think I know why..

The controller uses 2400 baud comms by default and the modules 5000 or 10000 baud. Right?

But this got me wondering, is this setup even possible? Can I get this to work with a controller firmware at 5000 baud or do I need the new controller?

Thanks..

stuartpittaway, 2023-05-23:
I added a 2K4 version of the 4.5 code (slower baud rate) to the last release, see if that one works with the old style controller.

floreseken, 2023-05-23:
I totally overlooked the 2K4 version for the1624, dumb of me..

stuartpittaway, 2023-05-23:
If you have it running reliably at 5000 baud, then stay at that speed, if you find you have communication issues then drop down to the 2400.

closed 2023-08-02 by stuartpittaway