stuartpittaway/diyBMSv4 #160: Consider adding a pdf-file for schematics

berrak, 2023-07-19:
I noted that the latest 4.9 design uses KiCad 7. However, only readily available on Windows, whereas on Linux and Mac, they still only have version six of KiCad. The old version can not open your version 7 KiCad design files or automatically convert any files.
A PDF of the schematics would be a welcomed addition, keep the interest up, and contribute later once KiCad is available on all non-Windows platforms. Cheers.

atanisoft, 2023-07-19:
@berrak I'm not sure where you are looking but KiCad 7.x has been available for Windows, Mac and Linux for a while now: https://www.kicad.org/download/

stuartpittaway, 2023-07-19:
@berrak if you check out the export folder, there should be automatically generated PDF documents for the circuit layout and schematics.

berrak, 2023-07-21:
Hi Stuart, thanks for hint about the export folder. Cheers.

berrak, 2023-07-21:
> @berrak I'm not sure where you are looking but KiCad 7.x has been available for Windows, Mac and Linux for a while now: https://www.kicad.org/download/

I was trying to say that many users still need to upgrade, as a Linux/Debian KiCad 7. x is not on the official repository. It is way out in 'testing,' which is years out. Not everybody has upgraded to the latest version of the macOS (not familiar with that OS). I like KiCad, but releasing new major updates (incompatible and without automatic tools for conversion) is annoying. Equally troublesome must external providers of footprints/schema suddenly need to be updated. Thus, is running the latest release necessary, and what are the fantastic benefits vs. version 6? Ultimately, it's up to Stuart, just my2c.

atanisoft, 2023-07-21:
> what are the fantastic benefits vs. version 6?

A number of bug fixes and improvements to the various tools mostly.

> as a Linux/Debian KiCad 7. x is not on the official repository. It is way out in 'testing,' which is years out.

This is one downside to many Linux distributions, a number of them stay on a given version "forever"... I'm on Gentoo myself and they have 7.0.2 as "stable" and 7.0.6 as "testing".

Due to these inconsistencies in uptake a number of tools are available as a Flatpak / Snap, including [KiCad](https://www.kicad.org/download/details/flatpak/). It's not exactly ideal but it does allow more rapid uptake when the distribution delays updates.

berrak, 2023-07-21:
> A number of bug fixes and improvements to the various tools mostly.
>
>
> This is one downside to many Linux distributions, a number of them stay on a given version "forever"... I'm on Gentoo myself and [...]

I did install the flatpak version - not ideal - but it works. I was wrong about not conversion v6 files, it's actually doing that. All good. Cheers.

atanisoft, 2023-07-21:
> I was wrong about not conversion v6 files, it's actually doing that.

Be warned on the conversion, it is a one-way conversion. Once the files are v7.x updated they will *NOT* open with v6.x.