stuartpittaway/diyBMSv4 #162: Communication wiring with (1 or more) 16S boards and controller

stuartpittaway, 2023-08-02:
This issue is purely documentation for connection of 16S monitoring boards and the DIYBMS v4.X controller.

Cabling is similar to the single cell monitoring modules.  

**DO NOT CONNECT TX TO TX OR RX TO RX - THE CORRECT WAY IS FROM TX TO RX - YOU CAN DAMAGE THE CONTROLLER IF MISTAKES ARE MADE**

_FOLLOWING DESCRIPTIONS ASSUME TOP DOWN VIEW ON THE CONNECTORS_

ON EACH 16S MONITORING BOARD, THERE IS A 4 PIN CONNECTOR WITH PINS MARKED ON SILKSCREEN AS:

- 3.3V
- RX
- TX1
- TX2

ON THE CONTROLLER

- Two pin JST connector marked RX
- Two pin JST connector marked TX  

**FOR CONNECTING A SINGLE 16S MONITORING BOARD**

CONTROL->TX->LEFT PIN = 16SBOARD->RX  (BLACK IN PHOTO)
CONTROL->TX->RIGHT PIN = 16SBOARD->3.3V (RED IN PHOTO)
16SBOARD->TX1 = CONTROL->RX->RIGHT PIN = (RED IN PHOTO)
16SBOARD->TX2 = CONTROL->RX->LEFT PIN = (BLACK IN PHOTO) 

**FOR CONNECTING MULTIPLE 16S MONITORING BOARDS**

CONTROL->TX->LEFT PIN = FIRST 16SBOARD->RX  (BLACK IN PHOTO)
CONTROL->TX->RIGHT PIN = FIRST 16SBOARD->3.3V (RED IN PHOTO)

FIRST 16SBOARD->TX1 = SECOND 16SBOARD->3.3V (ORANGE IN PHOTO)
FIRST 16SBOARD->TX2 = SECOND 16SBOARD->RX (ORANGE/WHITE IN PHOTO)

SECOND 16SBOARD->TX1 = CONTROL->RX->RIGHT PIN (RED IN PHOTO) 
SECOND 16SBOARD->TX2 = CONTROL->RX->LEFT PIN (BLACK IN PHOTO)

Photo below shows two 16S boards connected to a controller

[image]

chapulino, 2023-08-07:
I have tried the all in one and it works fine until it reaches 15 cells. When I connect number 16, the controller stays waiting for modules.
I started the tests with 16 cells and it didn't work, so I started by testing with 4 cells and increasing the number by two by two until I reached 14 cells. I have finally verified that up to 15 cells works fine, but when I go to 16 the problem appears.
In the openenergy forum you have recommended me:
  If you are looking at the code uint8_t queryAFE() is the function which counts the number of cells in use and then returns that to the controller.

If you want to compile the code to force 16 cells to test - change return cellCount; to return 16;
But unfortunately I don't know where to find that part of the code to be able to change it and compile it again in the ESP32

stuartpittaway, 2023-08-08:
Hi @chapulino.  Attached to this message is the firmware with a code for forcing 16S configuration.  This also uses the 10K baud/communication speed.  Unzip this and then write the code to the STM32 on the module as you have previously.
[module_fw_V490_10K_FORCED16S.zip](https://github.com/stuartpittaway/diyBMSv4/files/12288528/module_fw_V490_10K_FORCED16S.zip)

This will then allow us to determine if its a code or hardware problem.

chapulino, 2023-08-08:
thanks a lot
I'll definitely try it this afternoon and I'll report back to you.

stuartpittaway, 2023-08-08:
> I'll definitely try it this afternoon and I'll report back to you.

@chapulino I've found a bug for the 16S configuration - so please use the version on GITHUB which I updated today.  https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5796302484

chapulino, 2023-08-08:
work, thanks Stuart
now i tried the derivation of resistors

chapulino, 2023-08-09:
unfortunately the VREF450 version does not work as well as could be
expected, I have tried with 16 cells and I have wrong values in all of
them, well above the real value of the cells
[image: image.png]
This is the real value of the cells in an all in one VREF 4v

[image: image.png]
I don't know if I can help you with any test or measurement?

stuartpittaway, 2023-08-10:
Did you use the code specifically for the 4.5V reference?

chapulino, 2023-08-10:
this
Stuart Pittaway ***@***.***>
mar, 8 ago, 14:37 (hace 2 días)
para *stuartpittaway/diyBMSv4*, *Mention*, mí

I'll definitely try it this afternoon and I'll report back to you.

@chapulino <https://github.com/chapulino> I've found a bug for the 16S
configuration - so please use the version on GITHUB which I updated today.
https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5796302484

—

[image: image.png]

stuartpittaway, 2023-08-10:
None of the images are coming through from your emails, can you edit the GitHub issue directly please.

chapulino, 2023-08-10:
this 
[@chapulino](https://github.com/chapulino) I've found a bug for the 16S configuration - so please use the version on GITHUB which I updated today. https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5796302484

—
[image]

chapulino, 2023-08-10:
unfortunately the VREF450 version does not work as well as could be expected, I have tried with 16 cells and I have wrong values in all of them, well above the real value of the cells
image.png
This is the real value of the cells in an all in one VREF 4v

image.png
I don't know if I can help you with any test or measurement?

stuartpittaway, 2023-08-10:
Thank you for this information, can you confirm you have 4.50V on the test pin for the voltage reference.

This should just be a code bug.

chapulino, 2023-08-10:
yes i have 4.5V in the test pin

stuartpittaway, 2023-08-10:
Great, so it should just be a code/math change

chapulino, 2023-08-10:
I am going to send the All in one of lifepo4 (5) to the members of the group, since I have already verified them, I cannot send something that I am not sure works well. In the end they are very cheap, about €40 each set and I have a good time with this hobby and soldering components. this will broaden the tester base. I keep 3 for myself. I connected one of the bypass boards backwards and I have repaired a few resistors and maybe also the mosfets.
Another thing that happened to me is that in one of them, perhaps because of a jumper on one of the MAX pins, it did not give me a reading in a cell, I saw that the 100ohm resistor had burned. I am sending you a photo so you can laugh
[image]

stuartpittaway, 2023-08-10:
Ouch!

chapulino, 2023-08-11:
Coincidentally, in these two All in Ones that are going to go to 4.5v, I had not checked the voltage at TP5. One is correct, it shows 4.5 at TP% and at pin 1 of U3. The other has a voltage of 1.56 V. I leave you an image of the Digy-Key envelope and a photo of the assembly of the MCP1502T-45E
[image]
[image]

I have tried both firm 5K and 10k and the result is still not good I put an image
[image]

As for the one that does not have the 4.5V reference voltage, what should I check or replace?

chapulino, 2023-08-11:
Great Stuart 1 of All in One for 4,5v are ok 
[image]

stuartpittaway, 2023-08-11:
> As for the one that does not have the 4.5V reference voltage, what should I check or replace?

If one board works, and the other doesn't it has to be a soldering or component issue.

I'd start by re-soldering all the pins around U3/U6

chapulino, 2023-08-11:
i have actually resoldered everything including the max, i think this time JCBPCB did not do a good job of soldering

chapulino, 2023-08-11:
the second are ok also Thanks very much Stuart
Teh same expectacular stability and precisión.

[image]

stuartpittaway, 2023-08-11:
Time to get balancing the cells now then :-)

chapulino, 2023-08-17:
[image]
I add a schematic and close