stuartpittaway/diyBMSv4 #163: Addon Cuurent shunt board issue

anthonyevs, 2023-08-02:
Hi Stuart,
followed the install instructions and followed the soldering diagrams ok. Loaded the latest software onto the controller.
Got a 48v to 5vdc regulator fused for the power to the controller.
when I put the 5vdc supply onto the relay control positions the controller powers up, so i didnt think I needed to supply the controller with its own 5vdc supply as it was already powered thro the other route.
Now, with this wired in this way, all powers up, going into menu setup to activate the current shunt, set as DIY current monitor internal, saved settings, the icons pop up in tabs on top of the screen,
The problem is the current shunt isnt showing any voltage, I have put a multi meter onto the bat+ on the shunt and the neg tab and meter reads 53v which is correct as thats the battery voltage. The BMS reads 53v correctly.
So, why isnt the current monitor reading the battery voltage.
Do i need to give the controller its own 5vdc supply?
Any idea what may be the issue.
Thanks.

stuartpittaway, 2023-08-04:
> when I put the 5vdc supply onto the relay control positions the controller powers up

I don't follow this - what connections are you referring to?  The controller can only get power from one of 2 sources - either the ESP32 USB socket or the 2 pin 5V power connector on the board.

stuartpittaway, 2023-08-04:
> So, why isnt the current monitor reading the battery voltage.

I need a photo or wiring diagram of how you have connected it all up.

anthonyevs, 2023-08-04:
Hi Stu
so, yes, we have put 5vdc onto the 5v 4 port power connector, neg neg 5v 5v. This connector is powering the esp32 also.

I have not connected the usb on the esp32.
The shunt is recording ah out over the last 2 days, which is only the power that the bms and shunt is taking, nothing else is connected. 
I havent tried to charge or discharge the battery yet until this is resolved,
But also, i dont mind of there is an issue with the shunt recording the battery voltage, as that info is for me personally is of no use, i wont access that or use it in the rules to control the relays.
I hope this clears it up

stuartpittaway, 2023-08-07:
Hi @anthonyevs its really important that the voltage IS read by the shunt - otherwise is cannot do energy counting/metering.

This should be as simple as connecting the battery + terminal to the "+VBUS" connection on the controller.  You must ensure that the controller NEGATIVE is also connected to the negative terminal of the battery.

So run the +5V to the controller using a step down DC converter connected directly to the battery.

Avoid mains powered adapters for this.

stuartpittaway, 2023-08-11:
Hi @anthonyevs did you get this working ?

anthonyevs, 2023-08-11:
Not yet Stuart.
Ive double checked my wiring, thats all ok. Ive got 52v on a meter on the vbus to neg so confirming correct wiring.
Maybe the tiny chip isnt soldered pe4fectly on the shunt board. Can i use a meter to see if im getting voltage to the correct pin on the small chip on the shunt board? Ive had a quick look to try to trace the tracks but they dont seem too visible to positively identify where the 52v is going on the shunt board. Ill check the pcb wiring and boards diagrams later

stuartpittaway, 2023-08-11:
Here's a zoomed image.  

Can I check the wiring to the shunt as well....

* The battery negative is connected to the negative on the 5V input to the controller.
* The battery POSITIVE is connected to the "VOLT BATT+" terminal marked (J5)
* The shunt is connected to J3/J4 with the J3 terminal facing the battery.

The shunt should be connected in the POSITIVE line of the battery.

The battery voltage to the chip, should be on pin 8 (third down on right hand side).

[image]

MenoCore, 2023-08-15:
Hi Stuart, I would like to step in here as I experience obviously the same or a similar issue.
I only get 0-reading from the shunt tiles (see picture).
In the meantime, I soldered 2 boards and 3 addon boards as I was assuming soldering issues - no luck yet.
All connections double checked several times (Batt+ to J5, shunt to J5 and J3 / power supply step down to 5V properly attached). I am lacking of further ideas, maybe you can try to help - appreciate your kind feedback!

[image]

stuartpittaway, 2023-08-15:
Hi @MenoCore let's get this fixed!

What version of the controller do you have?

Can you get the usb serial console from the controller? It should report if it's detected the ina chip at boot up.

MenoCore, 2023-08-15:
Hi, wow you're fast :)
I have a 4.6 board and here the log - looks good to me.

https://pastebin.com/v7NHcBDC

###
I (5535) diybms-set: Read key (inf_apitoken)
I (5539) diybms-set: Read key (inf_orgid)
I (5542) curmon: Chip=INA229, Revision=1
I (5646) diybms: Onboard/internal current monitoring chip available
I (5646) curmon: RSHUNT=0.00166667, CURRENT_LSB=0.00005722, R_SHUNT_CAL=5000
W (5648) curmon: WARNING: True Max Current Measurable 24.58 Amps
I (5654) curmon: RECALC: RSHUNT=0.00166667, CURRENT_LSB=0.00004688, R_SHUNT_CAL=4096
I (5661) curmon: uint16 v=32768
I (5664) curmon: uint16 v=32768
D (5669) diybms: Setup RS485
D (5671) diybms: Configure RS485
D (5674) diybms-hal: SetOutputState 0=153
D (5677) diybms-hal: SetOutputState 1=153
D (5680) diybms-hal: SetOutputState 2=153
D (5684) diybms-hal: SetOutputState 3=153
I (5688) diybms: ** Controller changed state from PowerUp to Stabilizing **
###

stuartpittaway, 2023-08-15:
Thanks - that looks good, the chip is detected.

Do you have a photo of the setup?

Using a multimeter - what is the voltage between J5 (VOLT+ BATT) and the "GND" connection on the 5V controller input ?

MenoCore, 2023-08-15:
Ok... my setup looks like a mess - still in trial phase ;)

52.4V - fine.

[image]
[image]
[image]
[image]

stuartpittaway, 2023-08-15:
Ok - on the surface, that appears to be correctly wired up.  The voltage is also correct on that pin.

Can I ask you to try the latest version of code (pre-release) - its here...
https://github.com/stuartpittaway/diyBMSv4ESP32/suites/14946646590/artifacts/852244890

MenoCore, 2023-08-15:
Unfortunately not. Hard reset, new config. No shunt measurements...

               _          __ 
  _|  o       |_)  |\/|  (_  
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:7af23bb5063eac4fa519c8a12f570bd3c53d52ee compiled 2023-08-09T10:31:22.729Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (35) diybms-hal: Configure I2C
I (37) diybms-hal: Scanning i2c bus
I (44) diybms-hal: Found i2c device at address 0x21
I (55) diybms-hal: Found TCA6416A
I (59) diybms: ** Controller changed state from Unknown to PowerUp **
I (60) diybms-set: nvs_flash_init
I (62) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (65) diybms-hal: CAN driver started
I (75) diybms: LittleFS mounted, total=524288, used=139264
I (170) diybms: Mounting SD card
[   179][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   180][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
E (182) diybms-hal: Card mount failed

anthonyevs, 2023-08-15:
hi Stuart
> Here's a zoomed image.
>
> Can I check the wiring to the shunt as well....
>
> * The battery negative is connected to the negative on the 5V input to the controller.
> * The battery POSITIVE is connected to [...]

anthonyevs, 2023-08-15:
Hi Stuart, 
In my case ive found the fault, the current shunt board has blown the INA chip. It looks like there must have been abit of solder shorting the pins of the vbat+ middle pin on the IA chip and the GND pin on the INA next to the vbat+ pin (middle pin and the one below it) the top of the INA was blown off at this area.
Tonight Ive soldered a replacement shunt board onto the bms board. Double checked all pins on the INA and everywhere else for connection and shorting, all seems ok on this board.
A facebook group member kindly made the 2 boards up for me as i cannot do that level of soldering myself.
I have 6 spare INA chips (min qty order)  here (no boards), so will be trying to get this damaged one repaired with a new INA chip.
Going to test this out tomorrow.
Thanks Stuart.

stuartpittaway, 2023-08-16:
Thank you for letting me know your resolution @anthonyevs - its a shame you have a broken INA chip.  Those chips are really small and you definitely need to check for shorts on each pin using a magnifying glass or similar.

stuartpittaway, 2023-08-16:
@MenoCore the photos are not clear enough to look at the soldering on the chip - could you have a soldering issue on the INA chip?

stuartpittaway, 2023-08-16:
[image]
[image]

MenoCore, 2023-08-17:
Hi @stuartpittaway, really appreciate your continuous support!
Double checked everythig again and made some "macro" pics of the solder joints. Also traced the routes to the next components to make sure all are continuous and soldered properly + checked for bridges to neibor pins. All good.
I am a little frustrated and really sorry bothering you with this issue.

[image]
[image]

stuartpittaway, 2023-08-17:
So you have shown two different photos there one with the INA229 on the controller and the other with the add-on board.

You are NOT using both at the same time are you?

MenoCore, 2023-08-17:
That makes so much sense. I stupidly followed the build instruction for boards <=5.4 and never thought about that. LOL
Thank you, I will instantly try to rewire and let you know!

stuartpittaway, 2023-08-17:
> I will instantly try to rewire and let you know!

Just remove the board from the back of the controller!

MenoCore, 2023-08-17:
> 

What do I need to attach, just load and bat+? +VBUS can be ignored?

stuartpittaway, 2023-08-17:
The on-board current shunt uses the 4 pin header (screw terminals at top of board).

Its the same connections are the add-on board - look at the bottom silkscreen text and you can see 1 pin is not used.

+VBUS is for the voltage sensing - needs to go to the battery +

MenoCore, 2023-08-17:
Perfect, now all readings are there. Case closed. Thank you so much, again!
[image]

closed 2023-08-17 by stuartpittaway

anthonyevs, 2023-08-17:
just a heads up on my situation.
I attached a new shunt addon board to the bms board.
Checked with a meter for connections and shorts. all looks good.
Used a variable power supply and put 2v onto the vbatt + and board 5v gnd, all ok.
Increased voltage to 20v, all seems ok, no heating up or anything out of the way.
Going to connect it back to the battery tomorrow.
Thanks.

anthonyevs, 2023-08-18:
Hi Stuart,
can confirm the replacement addon shunt board is working perfectly.
Thank you.