stuartpittaway/diyBMSv4 #168: Run away balance I don't see that it is acting on cells

chapulino, 2023-08-16:
Stuart I continue with the tests, in this case I try to see how the Run Away system works (in the absence of an active balancer, only with the balancing plate).
But in principle I do not see that it acts by downloading any cell.
Maybe it's my mistake when configuring this function.
I start from a battery with cells in the range 3.85 v 3.70 v
[image]
[image]
[image]
Another detail, although this would be to open another issue, is that although there is no load, I have a Current reading of 0.15A
[image]

stuartpittaway, 2023-08-17:
Ok - the run away cell balance is meant to detect when the highest cell as "run away" from the others.

This is calculated by:

1. The highest cell has to be over the "cell minimum voltage" (specified on the configuration)
2. The difference of that cell has to be X millivolts over the AVERAGE of the other cells (X=specified on the configuration)

In your screenshot you have cell voltages of:
```
3.775
3.819
3.834
3.791
3.833
3.831
3.828
3.774
3.725
3.753
3.736
3.725
3.739
3.747
3.700
3.738
```

This gives an average cell voltage of 3772mV

The highest is cell 3 at 3834mV (this is over 3500 configuration value so valid)
The difference between the highest cell and the average is 62mV (3834 - 3772)

So in this scenario, **balancing on CELL 3 should be running**.

Note that the "balance board" is only detected on power up of the 16S board.  If you have plugged it in afterwards, balancing will not occur.

Can you raise a second issue if you have a problem with the current monitoring please.

chapulino, 2023-08-17:
Thank you, I have already been able to verify the operation, I was confused because I expected to see a download indication in the Cells monitor. I see that the balancing is done in the background.

Perfect
[image]

closed 2023-08-17 by chapulino