stuartpittaway/diyBMSv4 #170: AQY282SX relay is not longer made

stuartpittaway, 2023-09-24:
Update BOM to use

You can try part number C2691601
[Letex LT219-1](https://datasheet.lcsc.com/lcsc/2102221642_Letex-LT219-1_C2691601.pdf)

1 Form A(SPST-NO) 1.2V 60V 0.13Ω 1.3A AC,DC SOP-4-2.54mm Solid State Relays - MOS Output (PhotoMOS) ROHS

red0909, 2024-01-29:
i have killed two 282 on a v4.5 controller with a light load, they dont like spikes. 
on the older controller 4.4   AQY212GSZ - C129283 was used.
they switched for about a year every day a light load with no issues.

stuartpittaway, 2024-01-30:
I swapped over to the 212GSZ due to unavailability of the 282, the 282 has a better current handling capacity.

What are you driving with this relay output?

[AQY282SX](https://www.lcsc.com/product-detail/Solid-State-Relays-MOS-Output-PhotoMOS_PANASONIC-AQY282SX_C250827.html)
1 Form A(SPST-NO) 1.14V 60V 850mΩ 500mA AC,DC SOP-4-2.54mm Solid State Relays - MOS Output (PhotoMOS) ROHS

[AQY212GSZ](https://www.lcsc.com/product-detail/Solid-State-Relays-MOS-Output-PhotoMOS_PANASONIC-AQY212GSZ_C129283.html)
1 Form A(SPST-NO) 1.32V 60V 340mΩ 1A AC,DC SOP-4-2.54mm Solid State Relays - MOS Output (PhotoMOS) ROHS

red0909, 2024-01-30:
i switch fans on and off 
they are on a small step down regulator. the relay switches 48v to the regulator
changed todaa to old AQY212GSZ and now it is running without problems

maybe the 500ma was not enough for the step down small capacitor to charge it

stuartpittaway, 2024-01-31:
Ok, inductive loads are going to stress those digital relays out - the 212GSZ can handle 1Amp so would be better.

closed 2024-01-31 by stuartpittaway