stuartpittaway/diyBMSv4 #175: diyBMS cooperation with the 160S high-voltage battery

PiotrSin, 2023-12-02:
I have a li-ion battery from an electric car (car floor). Currently, this battery works with a Chinese BMS and a HYD 20 KTL-3PH inverter. 
It works well but lacks features available in diyBMS.
I have built a PCB to run diyBMS with such a battery, but the problem is that diyBMS only supports batteries up to 128S.
Please change the program code for 160S or more batteries. I will be happy to share my knowledge and present my project. 
I understand the difficulties with high voltage batteries. I know they are dangerous. 
I have the appropriate qualifications to work with high voltage.
thank you for your support

stuartpittaway, 2023-12-02:
Hi, how are you with compiling code? Using vscode?

I suggest that we could use the diybms alongside the existing BMS, particularly if you just want monitoring and reporting features.

Which diybms cell modules do you have?

PiotrSin, 2023-12-02:
I count on the author's help to change the code in BMS. I'm not good at programming.
Combining two different BMS systems does not seem to be a good solution
I have prepared 1 Controller 4.6 board and 10 ModuleV490-AllInOne boards

PiotrSin, 2023-12-02:
Li-lion 
2,8V > 4,2V
I use the range of 3V to 4.1V to increase durability

PiotrSin, 2023-12-02:
yes

stuartpittaway, 2023-12-02:
Ok, give me a day or two to review the code.

Might be worth while connecting up at least 1 of the modules to a bank for testing.

PiotrSin, 2023-12-04:
Thank you very much and I'll start testing.
Thanks also for the tip
How can I support this project or buy you a coffee :)

stuartpittaway, 2023-12-05:
> How can I support this project or buy you a coffee :)

You can also send beer tokens via Paypal - https://paypal.me/stuart2222

How are you going to control over current limits?

The DIYBMS doesn't include any current breakers that are often found in the Chinese BMS boards (MOSFETs) the reason is that the BMS can support very small and very large deployments, so didn't want to limit the flexibility.  The idea is that you would use a current shunt trip (via a controller relay) to trigger an external breaker.

PiotrSin, 2023-12-05:
This is exactly my plan, the BMS controls a relay to a high voltage contactor that disconnects the battery.
There are two more safeguards
- additional voltage comparators for each 16S section
- overcurrent protection
and additionally implemented temperature and fire protection with detection of gases from leaking batteries

For this set I will use a shunt developed by the user from this thread:
https://community.openenergymonitor.org/t/isolated-high-voltage-isolated-current-shunt-based-on-ads1115/24249

PiotrSin, 2023-12-05:
I have another question about programming the V490-AllInOne Module,

- I don't see this version in the AVR programmer tab, where can I find the correct version? / this link doesn't work https://github.com/stuartpittaway/diyBMSv4ESP32/tree/all-in-one/STM32All-In-One#readme

- where can I find the wiring diagram for programming cables from the ISP connector?

stuartpittaway, 2023-12-06:
You can't program the V490 boards directly from the controller - they use the STM32 chip not the ATMEL ones found on the older boards.

All the pre-compiled code is in the release section.

If you are NOT using the DIYBMS current sensor/monitor then you will also not be able to use the CANBUS integration (Pylontech/Victorn) if that is on your plan.

PiotrSin, 2023-12-09:
[image]

stuartpittaway, 2023-12-10:
The calibration settings has no affect on the v490 module, there shouldn't be any need for calibration as the design uses higher accuracy parts.

Vismantas-Braziulis, 2025-04-22:
PiotrSin, have you made it to work?