stuartpittaway/diyBMSv4 #178: Suggestion improve Cell runaway function

red0909, 2024-01-29:
at the moment it balances only one cell after an other, but it should balance all cells at same time if they are over the set voltage and deviation.

stuartpittaway, 2024-01-30:
The v4.9 board should balance any cell once it goes over the limit specified in "Bypass threshold voltage (mV)".

The run away cell function, is completely separate and tries to find high cell voltages before they reach this level.

red0909, 2024-01-30:
this bypass setting does not work if dynamic charge voltage is used. 
the cell voltages desnt go over the set bypass voltage because of the charging algorithm. the cell runaway does work but it balances only one cell. this is not enough if u have two cells low and 14 high.
so a change is needed to balance all cells those who are over the set start voltage and deviation

red0909, 2024-01-31:
this would be a nice feature if you can change this.