stuartpittaway/diyBMSv4 #183: Current Shunt addon board failed

anthonyevs, 2024-04-21:
Hi Stuart,
I have got 2 controllers using the addon current shunt board, one addon board failed a few weeks into its use, so I used the spare,
Its got to the time now where I need the addon board for a new project.
Sadly the INA chip blew up and off the board, I have nio idea why it did, as it was working for a few weeks before the unexpected catastrophic failure.
I have taken a photo of the addon board, i think the tracks may have ripped from the pcb when the chip blew itself apart. 
Have you got, or do you intend to get any more addon boards and sell them in the shop by any chance, or do you think this board may be able to get recycled.
Either way im loking for another 2.
If other patroens or members have any they dont require, would you consider sellng either 1 or 2 to me please, or offer tor repair the one I have,
Sadly i do not posess the skills or have the equipment to do the very intricate soldering to solder the INA chip onto the o=board, gosh its tiny.
Thansk all,
Best regards
Anth
[image]
ony.

stuartpittaway, 2024-04-22:
That looks fairly destroyed to me.  I don't have any plans to make any more of those boards as the new controller PCB has this built in, so no longer needed.

anthonyevs, 2024-04-22:
ok Stu, makes sense. Thing is, I have 2 good old type controllers with screens ready to go in new projects for my friends, so it`ll be a shame to write them off.
As I havent sourced anything from JLC pcb, is there any chance you could attach the correct files here that i would need to order up a min qty of these boards populated as they are all surface mount from them. All id need to do then is copy the files you attach into the JLC website, hopefully I wont make a mistake.
I know from the face group DIYBMS page another end user is looking for 2 as well, so if i ordered 5 he could have 2, ive got use for 2 then we would have 1 spare. 
Thanks Stu.

stuartpittaway, 2024-04-22:
I did have 30 of them for sale in my shop, but all gone now.

The jlcpcb files are all in GitHub, that's the same file I use to order for the shop