stuartpittaway/diyBMSv4 #186: All in One V490 Balancing function

Michi202020, 2024-06-24:
Hello Stuart,

(All in one version 490)

I would like to understand exactly how balancing works. Firstly, it is possible to set a global voltage, e.g. 4150mV. If this is exceeded, balancing is started. If the cell has e.g. 4000mV and the differential voltage is over 75mV (bank0 config), the highest cell balancing should be started but that doesn't happen. How should I understand that?

Best wishes, Michael.G

stuartpittaway, 2024-06-25:
Hi Michael,

There are 2 balance settings on the V490 boards which use the PASSIVE BALANCER BOARD - the first is a simple rule named "Bypass threshold voltage (mV) " if any cell goes over this voltage, then balancing will begin on that cell.

The second, is the "Run away balance cell".  This attempts to catch a single cell which is higher than the AVERAGE cell voltages of the other cells.    There are two settings, the first is an on/off feature - this function won't operate if the cell voltage is below this value ("Run away balance cell minimum voltage (mV)") the second value controls by how much a cell voltage needs to be above the AVERAGE to trigger balancing "Run away balance min. difference (mV)".

Therefore, in the screenshot example, if the average of the cells is 3425, and a cell is over 3445 - then that cell will start to balance.

[image]

If you find that balancing doesn't work at all, even though you have the passive balancer board fitted - then power off the V490 module (remove the 16way connector and the 2 pin connector) and reconnect.   The passive board is only detected on power up.

Michi202020, 2024-06-25:
Hello Stuart,
many thanks for your answer. I will check this.

Can you also see my question at the other side:
Discharge also a curve for amps go down #291

Many Thanks
Michael.G