stuartpittaway/diyBMSv4 #188: How to deal with more than 128 Cells?

Ipirk88, 2024-07-11:
Hello, i have connected 128 LiFepo4 Cells via 8 pcs "All In One Module V490" to an "Controller Board V4.6" and that all is connected via Can Bus to an Victron System.

Now i want to extend my Storrage to another 64 LiFepo4 Cells - total 192 Cells. 
How can i extend the System in that direction - i am fine when i have to use two Controller Boards - to IP adresses, ... 
But how to make shure that the Victron get all informations??

Next thing is the Shunt - i have now one 500A/50mV Shunt installed, but i want to use two 500A/50mV Shunts in parallel (i have limited space in hight - the 1000A Shunt will not fit.)??

So how to deal with that all? Any ideas?

Thanks so much for such a asome project!
Best regards,
Ingo Pirker

stuartpittaway, 2024-07-14:
You could compile the code to handle more than 128 cells, this has never been tested or required until now, but you can just change a single number up to 256.

Ipirk88, 2024-07-15:
Hey - thanks for the answer!
To compile the code should be not the issue. But what number should i change - can you maby describe this a bit more?!

And i will send you some good stuff to :-)
Paypal - https://paypal.me/stuart2222

Thank you so much!
BR 
Ingo Pirker

stuartpittaway, 2024-07-15:
Just this line..

https://github.com/stuartpittaway/diyBMSv4ESP32/blob/9a0a3ed26f2dc973d3faf3e2799a6688da3ab13e/ESPController/include/defines.h#L23

Ipirk88, 2024-07-15:
oh thats easy - thanks!
Will try and give feedback - but will take some time (shippment of the batterys will take 4 - 6 weeks)

closed 2025-05-21 by stuartpittaway

Ipirk88, 2025-05-21:
Sorry i forgot to give a info if it works - YES - verry good and verry stable: 

[image]

Thank you so much!

Ipirk88, 2025-05-21:
my old installation was 128Cells (16S 8P) - that was working perfect - so no issues there