stuartpittaway/diyBMSv4 #189: Maximum All in One V490

Michi202020, 2024-08-04:
Hello Staurt,

what is the maximum of All in one V490 on one Controller.
I have a LTO Storage and i need 40 pcs from this on one Controller can we make this or what can we do for realisation?

Best Regards and many Thanks
Michael.G

stuartpittaway, 2024-08-05:
What is the cell voltage and total pack voltage range you use?

Michi202020, 2024-08-06:
Hello Stuart,  my Lito pack have per cell 2,1 to 2,6v and its work good. Its round about 48v pack and this is What i used 40 Peace from the all in one v490 own creation. In Maximum there are 40x 12 cells= 480 cells

i Hope it Workshop or i Need 2 Controllers but Bothe Must Connect on one can Bus.

Best Regarda 
Michael.G

stuartpittaway, 2024-08-06:
Hi Michael.  Connecting two controllers to the same CANBUS isn't going to work, there are just too many cells for the DIYBMS system to handle in 1 controller.

Michi202020, 2024-08-07:
Hello Stuart, I think it also but how can I do this. It is possible that we can connect 2 or more controllers over the rs485 for calculation and push it on the can bus out. Or I must create a can bus hub what calculate all controllers and sent it on one can bus out. Thanks and best Regards 
Michael.G

Michi202020, 2024-08-07:
Hello Stiart I am again, can you say me what can bus adress do you programming in your code and can you say where is the pos innthe script for changing. I will also test on my Deye hybrid I mean I saw a mult bms list so that I can install mor bms on can bus.

thanks

Michi202020, 2024-08-09:
> Hi Michael. Connecting two controllers to the same CANBUS isn't going to work, there are just too many cells for the DIYBMS system to handle in 1 controller.

Hello Stuart, see my next question?
Best Regards 
Michael

stuartpittaway, 2024-08-09:
> Hello Stuart I am again, can you say me what can bus adress do you programming in your code and can you say where is the pos in script for changing. I will also test on my Deye hybrid I mean I saw a multi bms list so that I can install mor bms on can bus.

All the code for the can bus is in the file named canbus.cpp.

You still won't be able to put multiple controller on the canbus, it just doesn't work like that.

Michi202020, 2024-08-11:
Hy Stuart, ok I understand. In this case I make a can Bus hob and use 2 controller on one victron. I have create a controller withe 3 can bus and one rs485 and display for later Funktion. See [picture[](url)](
[image]
[image])

ZoMiGo, 2025-02-14:
> All the code for the can bus is in the file named canbus.cpp.
>
> You still won't be able to put multiple controller on the canbus, it just doesn't work like that.

Hallo Michael dein Code mäste das so ungefähr ausschauen bitte bedenke das das nur ein Beispiel ist 

/*
 ____  ____  _  _  ____  __  __  ___
(  _ \(_  _)( \/ )(  _ \(  \/  )/ __)
 )(_) )_)(_  \  /  ) _ < )    ( \__ \
(____/(____) (__) (____/(_/\/\_)(___/

(c) 2021-2023 Stuart Pittaway

Dieser Code kommuniziert mit VICTRON CERBO GX über CAN-Bus @ 500kbps mit 11-Bit-Adressen.
Er unterstützt die Victron CAN-Bus BMS Nachrichten.

DEMO Zusätzliche Erweiterungen by Trajilovic Goran
- Master-Slave-Erkennung über CAN-Bus
- Master aggregiert Daten von Slaves und sendet sie an Victron
*/

#define USE_ESP_IDF_LOG 1
static constexpr const char *const TAG = "diybms-victron";

#include "victron_canbus.h"

// ==================== Master-Slave Konfiguration ====================
#define MAX_SLAVES 10
#define MASTER_ID 1 // Kleinste ID wird Master
bool isMaster = false;
unsigned long lastMasterHeartbeat = 0;

struct SlaveData
{
    uint16_t voltage;
    int16_t current;
    int16_t temperature;
};

SlaveData slaveNodes[MAX_SLAVES];
bool slaveOnline[MAX_SLAVES] = {false};

// ==================== Slave sendet Daten an Master ====================
void send_slave_data()
{
    SlaveData data;
    data.voltage = rules.lowestBankVoltage / 100;
    data.current = mysettings.dischargecurrent;
    data.temperature = rules.highestExternalTemp;

    uint16_t can_id = 0x200 + mysettings.slave_id;
    send_canbus_message(can_id, (uint8_t *)&data, sizeof(SlaveData));
}

// ==================== Master verarbeitet Slave-Daten ====================
void process_slave_data(uint32_t can_id, uint8_t *data, uint8_t length)
{
    if (can_id >= 0x200 && can_id < 0x200 + MAX_SLAVES)
    {
        uint8_t slaveId = can_id - 0x200;
        memcpy(&slaveNodes[slaveId], data, sizeof(SlaveData));
        slaveOnline[slaveId] = true;
    }
}

// ==================== Master sendet Daten an Victron ====================
void send_aggregated_data_to_victron()
{
    struct VictronData
[... 153 lines cut ...]
            data.byte0 |= (rules.ruleOutcome(Rule::ModuleOverTemperatureExternal) ? 0x40 : 0x80);
        }
    }

    send_canbus_message(0x35a, (uint8_t *)&data, sizeof(data35a));
}

// ==================== Haupt-Loop ====================
void loop()
{
    check_master_status();

    if (!isMaster)
    {
        send_slave_data();
    }

    if (isMaster)
    {
        send_aggregated_data_to_victron();
    }

    send_heartbeat();
    delay(5000);
}

ZoMiGo, 2025-02-14:
` /*
 ____  ____  _  _  ____  __  __  ___
(  _ \(_  _)( \/ )(  _ \(  \/  )/ __)
 )(_) )_)(_  \  /  ) _ < )    ( \__ \
(____/(____) (__) (____/(_/\/\_)(___/

(c) 2021-2023 Stuart Pittaway

Dieser Code kommuniziert mit VICTRON CERBO GX über CAN-Bus @ 500kbps mit 11-Bit-Adressen.
Er unterstützt die Victron CAN-Bus BMS Nachrichten.

DEMO Zusätzliche Erweiterungen by Trajilovic Goran:
- Master-Slave-Erkennung über CAN-Bus
- Master aggregiert Daten von Slaves und sendet sie an Victron
*/

#define USE_ESP_IDF_LOG 1
static constexpr const char *const TAG = "diybms-victron";

#include "victron_canbus.h"

// ==================== Master-Slave Konfiguration ====================
#define MAX_SLAVES 10
#define MASTER_ID 1 // Kleinste ID wird Master
bool isMaster = false;
unsigned long lastMasterHeartbeat = 0;

struct SlaveData
{
    uint16_t voltage;
    int16_t current;
    int16_t temperature;
};

SlaveData slaveNodes[MAX_SLAVES];
bool slaveOnline[MAX_SLAVES] = {false};

// ==================== Slave sendet Daten an Master ====================
void send_slave_data()
{
    SlaveData data;
    data.voltage = rules.lowestBankVoltage / 100;
    data.current = mysettings.dischargecurrent;
    data.temperature = rules.highestExternalTemp;

    uint16_t can_id = 0x200 + mysettings.slave_id;
    send_canbus_message(can_id, (uint8_t *)&data, sizeof(SlaveData));
}

// ==================== Master verarbeitet Slave-Daten ====================
void process_slave_data(uint32_t can_id, uint8_t *data, uint8_t length)
{
    if (can_id >= 0x200 && can_id < 0x200 + MAX_SLAVES)
    {
        uint8_t slaveId = can_id - 0x200;
        memcpy(&slaveNodes[slaveId], data, sizeof(SlaveData));
        slaveOnline[slaveId] = true;
    }
}

// ==================== Master sendet Daten an Victron ====================
void send_aggregated_data_to_victron()
{
    struct VictronData
    {
        int16_t voltage;
        int16_t current;
        int16_t temperature;
    };

[... 147 lines cut ...]
            data.byte0 |= (rules.ruleOutcome(Rule::ModuleOverTemperatureExternal) ? 0x40 : 0x80);
        }
    }

    send_canbus_message(0x35a, (uint8_t *)&data, sizeof(data35a));
}

// ==================== Haupt-Loop ====================
void loop()
{
    check_master_status();

    if (!isMaster)
    {
        send_slave_data();
    }

    if (isMaster)
    {
        send_aggregated_data_to_victron();
    }

    send_heartbeat();
    delay(5000);
} `