stuartpittaway/diyBMSv4 #193: All-in-on zero volt

botaneta, 2024-09-15:
Hello, 
I have a controller and an all-in-one for testing. I have never had this error occur with the 36s All-in-one modules. On some occasions the All-in-one module reports zero_volt, restarting the module solves it. 
Can the controller send a command to reset the All-in-one module that has zero_volt?
Thank you.

stuartpittaway, 2024-09-16:
Does this only happen on this one board? Nobody else has reported this problem.

I also see that your connectors are not fully inserted, there shouldn't be any gap when fully plugged in

botaneta, 2024-09-17:
Yes, it happens only on this board. I'll check the connector.
Thank you

stuartpittaway, 2025-05-21:
As this issue is old, I'll close it - please reopen if you still have problems.

closed 2025-05-21 by stuartpittaway