stuartpittaway/diyBMSv4 #194: AH counting IN when BMS is isolated from inverter and inverter is powered off

anthonyevs, 2024-10-10:
Hi all, 
i have a slight issue with the new type all in one boards and new controller in my brothers system.
With the bms working on its own, EV200 isolators / contacotrs switched off to the inverter, and the inverter powered down, and off, no other power source connected, the BMS somehow is counting IN ah.
in History, since 5pm last night the AH in has gone from 
AH in 442520 to 347805
yet the voltage has gone from 47.92v to 47.8v 
So, the voltage has dropped (which you would edpect as the bms is powering itself)
yet the AH in has increased.
Strange,
What is happening here?
What this has done also has increased the SOC from 86- 89%
Over time the SOC now has shifted away from where it really is, so it now becomes an hindrance as the SOC is always shifting and cannot be trusted. 
Any help greatly appreciated. 
Thanks.
Anthony.

stuartpittaway, 2024-10-10:
With the contactors open, the current shunt is floating, possibly giving the false readings.

What was the current reading during this time?

anthonyevs, 2024-10-10:
Hi Stu,
the current reading in history all thro the timeline was 0.122 amps every 1/2 hr interval as recorded.
The bms gets its live and negative from the bms side of the contactors, so when the contactir on the live and negative are open, the bms still has power.

stuartpittaway, 2024-10-10:
So isn't this current reading the power to run the BMS module and controller?

anthonyevs, 2024-10-10:
it is, but surely, as its taking power out of the battery, it should be counting the AH OUT not in, and hence reducing the battery voltage, the battery SOC,
As it is its counting IN capacity into the battery and then messing up the battery SOC. This should not be, surely.