stuartpittaway/diyBMSv4 #196: DIYBMS All-in-one 16S

romeo0386, 2025-03-04:
Good afternoon Stuart, I have a communication problem with the Victron system and DIY BMS, but with the 16S1P module system I can connect to the Cerbo and the only thing that gives me is an error on the Muntiplus 2, it tells me that the battery is low and the inverter turns off.

[image]

stuartpittaway, 2025-03-06:
Hello, this is very unusual.  You can see that the DIYBMS is sending the correct voltage and state of charge to Victron.

I expect there is a setting in the MultiPlus or Cerbo which needs changing, but I don't have a Victron system to advise you on.

Are all the firmware updates done on Victron?

romeo0386, 2025-03-06:
Good afternoon Stuart, thank you for answering the message, but I have
already solved the problem. The problem was because I did not put the
temperature sensor on the 162 1p module.

closed 2025-03-07 by stuartpittaway

romeo0386, 2025-03-07:
Good afternoon Stuart, I wanted to ask you what day the controller version
v4.61 will be available or if you have the files to be able to buy them at
jlcpcb